vvv-school/vvv18 #2: Quick unpacking tip for Linux users

fbottarel, 2018-01-30:
For anybody @vvv-school/vvv18-students installing the **VVV18 Virtual Machine** under Linux.
The zip file you download will contain **three files** with extensions .zip, .z01, z02. If you're a Linux user, there's the chance the archive manager bundled with your distro (e.g. FileRoller) won't be able to handle such archive. Quick fix:
- install 7zip for Linux `sudo apt-get install p7zip-full `
- unpack the archive you download from Figshare `7z e [archive name].zip`
- at this point, you will end up with three files `[archive name].zip [archive name].z01 [archive name].z02 `. Apparently 7zip cannot link the three if they are named as such though
- rename the .z01 and .z02 files as `[archive name].zip.001 [archive name].zip.002` and [archive name].zip as `[archive name].zip.003`
- run `7z e [archive name].zip.001` to have 7zip merge the archive and extract your .ova file. 

Enjoy. Remember you can always buy me a drink if this helped you :beer:

diegoferigo, 2018-01-30:
Thank you @fbottarel for the how-to. I'm providing more details about the steps for tackling possible pitfalls you might encounter (on Linux systems).

After the three files from figshare have been correctly downloaded:

#### 1. Check the md5sums of the files
```bash
$ md5sum VVV18VM.z01 
ff2e56233d5f7f9a3127ac12d94c66e5  VVV18VM.z01
$ md5sum VVV18VM.z02 
4b47900a7ea0cd7b21ec73fe4fc459c9  VVV18VM.z02
md5sum VVV18VM.zip 
9e8119105ddd57f27e6576ee382f59b8  VVV18VM.zip
```

If the sums does not match, the download was not successful and the files are corrupted. Download them again. These sums are also available in the figshare page by clicking the `View` link that appears hovering the file.

#### 2a. Extract the files using `zip`

The archives have been creating using `zip 3`. The proper way to merge and extract the files is the following: 

```bash
# Merge the files
zip -s 0 VVV18VM.zip --out VVV18VM_merged.zip
# Extract the archive
unzip VVV18VM_merged.zip
```

If you encounter any problem when unzipping the merged archive, it could mean that the file contains some small errors. Execute:

```bash
# Fix the merged archive (this will perform another copy of the file)
zip -F VVV18VM_merged.zip --out VVV18VM_merged_fixed.zip
# Extract the fixed archive
unzip VVV18VM_merged_fixed.zip
```

#### 2b. Extracting the files using `7z`

`7z` is a valid alternative. Follow the instructions @fbottarel provided.

diegoferigo, 2018-01-30:
Considering the inconsistencies among operating systems using the `zip` format, we decided to upload a new `rar` split archive. This format handles in a better way split archives.

Here the updated instructions:

#### Install the required packages

```bash
$ sudo apt install rar
```

#### Validate the md5sum of the `rar` files

```
$ md5sum VVV18VM.*.rar
e9f6f14b540b4c450fea468db2dfc32b  VVV18VM.part1.rar
0573401ce93178da566a3ab51b6d71ea  VVV18VM.part2.rar
e7f7733a4ae148313efe3c73b10fef36  VVV18VM.part3.rar

```

#### Extract the split archive

```bash
$ rar x VVV18VM.part1.rar
```

#### Validate the md5sum of the resulting `ova` file

```bash
$ md5sum iCub\ Facility\ -\ Ubuntu\ 16.04\ LTS.ova
727ae6484a5884144bd3770024bbe90d  iCub Facility - Ubuntu 16.04 LTS.ova
```

bmagyar, 2018-01-30:
... and for those who happen to be using the same version of Ubuntu as the
VM I'd humbly recommend installing everything locally for a better runtime
experience.

pattacini, 2018-01-30:
@bmagyar well, actually our advice is exactly the opposite: don't go with your own installation but rather use the VM. Providing support for own installations is a burden we cannot afford with the fairly tight schedule we'll have during the school.

diegoferigo, 2018-01-30:
@bmagyar We are aware that virtualbox has performance issues, probably due to a regression happened in the latest versions. I had really nice performance using `VMware Player`, which can be downloaded for free from the website. For personal use it doesn't ask for any license. With the `open-vm-tools` package that can be found in the Ubuntu repositories the screen resolution can be increased.

If any of you on you personal machine will face performance issues running virtualbox, you're encouraged to give `VMware Player` a try.

bmagyar, 2018-01-30:
Point taken :) Sorry for the noise guys, feel free to remove these comments from this thread to avoid confusing students!

closed 2018-02-06 by pattacini