vvv-school/vvv18 #11: Assignment on YARP motor control

Nicogene, 2018-02-07:
Dear @vvv-school/vvv18-students, now that you are `YARP` experts it's time to practice what you learnt in these days 💪 

The assignment should be easy-peasy, by the way @vvv-school/vvv18-teachers are here to help you 😉 

To start the assignment please click here: https://classroom.github.com/a/hFAwZloC 🚀 

Happy coding 🤖

pattacini, 2018-02-07:
Hey @vvv-school/vvv18-students 

As we said during the assignment, to properly debug your code it is vital to print out messages during the execution of your instructions to catch what is going on. To this purpose, you may use `yDebug()`, `yWarning()`, `yInfo()` directives and the like.

For example, do:
```c++
yDebug()<<"we reached this point we shouldn't reach :(";
```
The `yDebug()` directive does send the message to the `yarplogger` where it is conveniently displayed.

For a glitch in the preparation of the assignment 🐞, the xml application that launches your binaries is not shipped with the correct option.

Therefore, please apply the following [**patch**][1] to your `app/scripts/assignment_motor-control.xml` application and do `make install`.

#### Closing remark
The `yarplogger` does not shut down when you click on the proper button within the `yarpmanager` because of the command line option `--no_stop`. If you get rid of it, `yarplogger` will close as expected. That option serves purposely to let the logger catch the terminal messages upon closing the other modules 😏 

[1]: https://github.com/vvv-school/assignment_motor-control/commit/fa978226152e49277c7df27787b97bffe679b0e7

Nicogene, 2018-02-07:
Hey @vvv-school/vvv18-students,
 Maybe the `README` was not clear but the trigger sends a a signal to the client that has not to be interpreted. In some solution I saw something like this:
❌ 
``` c++
Bottle *bIn = inputPort.read();
triggered = bIn->get(0).asBool();
```
 but what it is sent from the `trigger` is not a bool.

Just  call:
✅ 
```
Bottle *bIn = inputPort.read();
triggered = true;
```

The read has to be invoked in the blocking way, so as soon as it reads something the `triggered` flag has to be set to true, no matter what it is inside the `Bottle`

closed 2018-02-08 by pattacini