vvv-school/vvv18 #28: Bugfix in tutorial_impedance_control

prashanthr05, 2018-02-10:
The tutorial_impedance_control module name seems to have a bug when called through the yarpmanager. To fix this open the  
`tutorial_impedance-control-{your_name}/app/scripts/tutorial_impedance-control-app.xml` with your favorite text editor and find the tag 
```
<module>
    <name>tutorial_impedance-control</name>
```

and change it to 

```
<module>
    <name>tutorial_impedance_control</name>
```

Save the file, go to the build directory and do a `make install`. Reopen `yarpmanager` and try running the module again. Should work fine!

prashanthr05, 2018-02-10:
@vvv-school/vvv-students

pattacini, 2018-02-10:
Or even better, change the project name from `tutorial_impedance_control` to `tutorial_impedance-control` into the CMakeLists.txt file.

closed 2018-02-10 by pattacini