vvv-school/vvv18 #41: Gradebook update

science-code, 2018-02-13:
Dear Ugo @pattacini,

Would it be possible to launch the grader for the following two assignments?

Gradebook in YARP Middleware
Gradebook in Robot Dynamics

Thank you and kind regards,
Mihai

pattacini, 2018-02-13:
Of course, gimme just the necessary time! 👍

marco-monforte, 2018-02-13:
I would ask the same for the Event-based Vision gradebook :)

Thanks!

pattacini, 2018-02-13:
Sure @marco-monforte 😉

pattacini, 2018-02-13:
Done!

science-code, 2018-02-13:
@pattacini 
Thank you!
However, I still have a problem with the assignment on YARP Middleware: I get 14/14 on my machine, but only 10/14 on the remote machine...

pattacini, 2018-02-13:
Ehm, do you know that there's a decay function that reduces the max points on the grading machine depending on the delay of the submission?

Just kidding! Actually, I've seen this weird behavior on the physical machine and let me call for @Nicogene who could perhaps patch the smoke test in this respect.

science-code, 2018-02-13:
@pattacini, @Nicogene Thank you! I will be waiting for an update from your side in this case.

Nicogene, 2018-02-13:
Sorry guys my bad, I made some changes in `assignment_motor-control` for the next year and I didn't know that the repo was still used by the grading machine!
Now should work

pattacini, 2018-02-13:
Grading's underway. Stay tuned.

pattacini, 2018-02-13:
Done. Check it out ✨

gonzalezJohnas, 2018-02-13:
Dear @pattacini ,
Could you run please the grader for : 
Yarp vision
Yarp Dynamics

Grazie mile

pattacini, 2018-02-13:
Sure @gonzalezJohnas

pattacini, 2018-02-13:
Done.

closed 2018-02-13 by pattacini