vvv-school/vvv18 #56: Grand Challenge Ranking

pattacini, 2018-02-17:
Probably you're already safe at home or spending the weekend somewhere to have a break as well as a quick recovery from the tough days 😉 but let me recap here below the final ranking of the grand challenge!

| | Team | Repository |
| :---: | :--- | :--- |
| :one: | 📙 @vvv-school/vvv18-team-orange 🥇🏆 | [`vvv18-demo-team-orange`](https://github.com/vvv-school/vvv18-demo-team-orange) |
| :two: | 📗 @vvv-school/vvv18-team-green 🏅 | [`vvv18-demo-team-green`](https://github.com/vvv-school/vvv18-demo-team-green) |
| :three: | 📘 @vvv-school/vvv18-team-blue 🎖 | [`vvv18-demo-team-blue`](https://github.com/vvv-school/vvv18-demo-team-blue) |

📙 What counted in favor of the Oranges in the final fight against the Greens was that they eventually managed to show the state machine at work continuously and they also delivered a pretty nicely documented package! The object recognition was their downside: it did not run but this did not prevent the robot from localizing and pointing at objects, at least.

📗 On the other hand, the vision system of the Greens functioned amazingly well! It's a pity that teachers were unable to watch the Dab during the contest!

📘 The Blues ranked last only because they got problems while running the code on the robot ad we didn't see that much, although we perfectly know that troubleshooting under pressure is not ideal!

📝 Remember: never ever change the robot and the system you used during the tests a couple of minutes before the demo!

That being said, the assignments and the contest have been only the excuse to push you to apply what you have learned during the courses and we are super happy with the results you all achieved! We thank you again for your great commitment ✨ 

👉 Don't forget to fill in the questionnaire, which will help us improve the next edition of the school!

Finally, if you have any pictures of the last day you want to share, feel free to post them herein. We are going to share ours soon.

pattacini, 2018-02-17:
cc @vvv-school/vvv18-students @vvv-school/vvv18-teachers

science-code, 2018-02-17:
The segment of green team's demo with Purpley: https://photos.app.goo.gl/f2OfLlSQSWVmAtMt2

pattacini, 2018-02-17:
One of the best pointing I've ever seen! 😃 
Thanks @science-code for sharing.

lornat75, 2018-02-17:
The podium!

[image]

[image]

[image]

closed 2018-02-19 by pattacini