vvv-school/vvv18 #57: VVVWinterSchool org

pattacini, 2018-02-19:
Hi @ninja777 

Hope you're doing fine! I've just seen this new organization https://github.com/VVVWinterSchool where your hands-on are being copied in.

That's great, but would you be willing to change the name of the organization into something that wouldn't clash/overlap in any way with our vvv-school? 😉

ninja777, 2018-02-19:
Sure. Sorry didn't realize that as I wanted to get the backup before everything is archived :)
These repositories are for future reference only. Let me know if you would prefer them to be private.

pattacini, 2018-02-19:
Even `WinterSchoolHumanoidProgramming` might be somewhat confusing, in the sense that while googling our school one can end up in your org. What about suffixing your org with something related to your GitHub username or anything equivalent? 😄 

Then, your repositories can remain public, as you like.

ninja777, 2018-02-19:
I hope this name will work `NotTheVVVWinterSchoolNinja777`. This is inspired by Elon Musk's tweet [about flamethrower](https://twitter.com/elonmusk/status/959555569953660928) but in good spirit :)

I have also added the description as `This is a personal project of repos from VVV winter school 2018. If you are searching for official repos, please visit https://github.com/vvv-school` which will help future students to p[oint to the right place.

pattacini, 2018-02-19:
That's perfect! 👍 
Thanks @ninja777!

closed 2018-02-19 by pattacini