vvv-school/vvv18 #58: When I cloned the all codes, I find most of codes cannot be run correctly,  and only a few can be run correctly.

YANGGANG1971, 2019-11-11:
Day 1 - Tuesday 06/02
Welcome
Git Basics
•	🎖 assignment_git   		Failed: Test.sh cannot be run.
YARP
•	🔨 tutorial_yarpmanager	 Failed
•	🔨 tutorial_yarp-ports     	Failed: 	
•	 	added Port inPort; inport.open(“/in”);
•	🔨 tutorial_yarp-producer-consumer	 Succeeded
•	🔨 tutorial_yarp-rpc 	Failed
  
Day 2 - Wednesday 07/02
YARP
•	🔨 tutorial_RFModule-simple     		Succeeded
•	🔨 tutorial_joint-interface       		Failed
•	🎖 assignment_motor-control 		Failed 
  
ROS
•	🔨 tutorial_ros          			Failed
Day 3 - Thursday 08/02
Robot Kinematics
•	🔨 tutorial_inverse-kinematics      	Failed
•	🎖 assignment_inverse-kinematics      	Failed
•	🔨 tutorial_cartesian-interface          	Failed
•	🔨 tutorial_gaze-interface                	Failed
•	🎖 assignment_make-it-roll                	Failed
Day 4 - Friday 09/02
Robot Vision
•	🔨 tutorial_find-wally            		Failed
•	🔨 tutorial_yarp-opencv               		Failed
•	🎖 assignment_closest-blob           		Failed
Day 5 - Saturday 10/02
Robot Dynamics
•	🔨 tutorial_impedance               		Failed
•	🎖 assignment_computed-torque        	Failed
•	🔨 tutorial_impedance-simulink             	Failed
•	🎖 assignment_computed-torque-simulink   Failed
Day 7 - Monday 12/02
Event Based Vision
•	🔨 tutorial_event-driven-framework  	Failed
•	🔨 tutorial_event-handling                	Failed
•	🎖 assignment_event-saccadic-suppression        Failed
•	🎖 assignment_event-spiking-attention             Failed
Day 8 - Tuesday 13/02
Machine Learning & Deep Learning
•	🔨 tutorial_dl-tuning                         Succeeded 
•	🎖 assignment_dl-tuning                       Failed
•	🔨 tutorial_dl-deployment              
Prerequisite
	System check
		Linux: Unbuntu 16.04
		Yarp:   YARP version 3.2.1+5-20190926.5+gitfe0de91cc
		CMake: 3.5.1
Main questions:
1.	While running the following command line, it seems something wrong as follows. Once I type ./tesh.sh, similar errors also will occur.
$ git clone https://github.com/vvv-school/tutorial_cartesian-interface.git
$ cd tutorial_cartesian-interface/smoke-test
$ git checkout vvv18
$ ./test.sh
[image]
Figure: Error while running “./test.sh”

2.	When I want to run “cmake ../”, always warning will occur (see the following)—The YARP_MODULE_PATH variable is deprecated. Does it affect the executable after compiled? Most tutorial program can be run correctly. is it relevant?
 
[image]
Figure: Warning while running “cmake ../”

When I type “ccmake ../”, it will be  the following configuration. Is it correct?
[image]
Figure: Configuration while running “ccmake ../”

3.	Normally, when I run the command line “git clone …”, then I will get some codes from Github. But when I compiled using the “cmake ../” command and get the executable, most programs cannot get the ideal results.  So I noticed that some projects are incomplete (not only for assignment tasks). Could you mind giving the complete codes, which can run well? Or could you tell me how to get the correct codes to revise the code written by myself?

pattacini, 2019-11-11:
Hi @YANGGANG1971 

I'm afraid that [YARP 3.2.1][1] is too recent to be used along with **Ubuntu 16.04**.
My warm recommendation is to employ instead the [VM we made available][2] at the time of VVV18, which comes with all the necessary software components preinstalled.

However, I've also made an attempt to run the following assignments:

- https://github.com/vvv-school/tutorial_cartesian-interface
- https://github.com/vvv-school/assignment_make-it-roll

within a system recreated using this [Docker recipe][3]. Both ran successfully 👍 

Interestingly enough, you could consider getting a copy of such an environment within your browser by relying on [Gitpod][4].

Just click on this badge

[image]

sign up for the service using your GitHub credentials and then follow the instructions outlined in [this post][5] to have the graphical desktop accessible in the browser. From that point, you can start cloning the assignments.

Coming down to your questions:
1. The screenshot is way too small to be readable. Anyway, bear in mind the considerations I've summarized above.
1. Deprecation of `YARP_MODULE_PATH` doesn't prevent you from building the binaries.
1. Assignments are meant to be not fully operational on purpose since students are required to turn in their solutions. Solutions are hosted in private repositories that cannot be disclosed, obviously. To this end, the mechanism of smoke-tests will serve to verify your proposal. My suggestion is that you try out first your approach and only then you can get back to us and we will possibly give you hints/correct solutions.

Hope this helps out.

[1]: https://github.com/robotology/yarp/releases/tag/v3.2.1
[2]: https://github.com/vvv-school/vvv18/wiki/Before-arriving-at-VVV#prepare-your-system
[3]: https://github.com/pattacini/dockerfiles/blob/master/technical-evaluation-stack/gitpod/Dockerfile
[4]: https://gitpod.io
[5]: https://education.github.community/t/the-benefits-of-integrating-docker-and-cloud-ides-in-the-github-classroom-workflow/40724

YANGGANG1971, 2019-11-12:
I appreciate you helping me a lot.
I try to do it as you recommended. First download iCub Facility - Ubuntu (x64) VVV VM (https://figshare.com/articles/iCub_Ubuntu_16_04x64_Virtual_Machine/4815046/4).
The yarp version is 2.3.71.1.
But, while running the following codes,
$ yarp namespace /YANGGANG1971
$ yarpserver --write
$ yarp check
$ git clone https://github.com/vvv-school/tutorial_cartesian-interface.git
$ cd tutorial_cartesian-interface/smoke-test
$ git checkout vvv18
$ ./test.sh
the errors still occurred (See the following figure).  
[image]

pattacini, 2019-11-12:
Now that I can read the log, I can see that the problem is caused by the fact that the binary `rtf-testrunner` has been renamed in `robottestingframework-testrunner` in the latest releases.

That said, I've spotted a bug in the workflow that makes smoke-testing unable to rely on the correct version of the scripts used in the process. See github.com/vvv-school/vvv-school.github.io/issues/9 for more details.

One possible workaround could be:
1. Launch the `./test.sh --get-helpers` script at least once. It will fail.
1. `cd build/helpers`
1. `git checkout vvv18`
1. Go up again: `cd ../..`
1. Launch `./test.sh`

I will look into fixing this bug as soon as I'll have some time.

YANGGANG1971, 2019-11-13:
It seems that the workaround does not work as follows.
1.	Launch the ./test.sh --get-helpers script at least once. It will fail.
[image]
2.	cd build/helpers
3.	git checkout vvv18
[image]
4.	Go up again: cd ../..
5.	Launch ./test.sh
[image]

YANGGANG1971, 2019-11-13:
Also, I found the YARP version incompatible. I altered the “find_package(YARP 2.3.71.1 REQUIRED)”as “find_package(YARP 2.3.71.1 REQUIRED)”. But some new errors also occurred.
1.	Launch the ./test.sh --get-helpers script at least once. It will fail.
[image]
2.	cd build/helpers
3.	git checkout vvv18
[image]
4.	Go up again: cd ../..
5.	Launch ./test.sh
[image]

pattacini, 2019-11-13:
Likely, you missed to check out the `vvv18` tag once you cloned `tutorial_cartesian-interface`.

YANGGANG1971, 2019-11-13:
It seems that I have run git checkout vvv18. My codes are as follows,
>>git clone https://github.com/vvv-school/tutorial_cartesian-interface.git
>>cd tutorial_cartesian-interface/smoke-test
>> git checkout vvv18
[image]
>> ./test.sh --get-helpers 
[image]
>> cd build/helpers
>> git checkout vvv18
[image]
>> cd ../..cd
>> ./test.sh
[image]

pattacini, 2019-11-13:
Right, there's another error, which stems from the fact that the tag `vvv18` of helpers hasn't been retrieved correctly from GitHub.

Now, the sequence again:
```sh
$ git clone https://github.com/vvv-school/tutorial_cartesian-interface.git
$ cd tutorial_cartesian-interface/smoke-test
$ git checkout vvv18
$ ./test.sh --get-helpers
(it will fail)
$ cd build/helpers
$ git pull --tags
$ git checkout vvv18
$ git ../..
$ ./test.sh
```

Consider that I'm unable to try it out on the VM, hence there might be other problems. With this in mind, it'd be better to go with Gitpod.

YANGGANG1971, 2019-11-13:
You are right (only change last second command "git" as "cd", it seems ok).
Thank you. 
I prefer to choose the Gitpod to test the code and it really is impressive and useful.
But, finally, I need to test it on real iCub.

Can the Gitpod control the real robot (not only simulation)? How to do that?

Sometimes, when I wanted to test the codes on Gitpod, I found "failed to connect server".
I guess that the conflict caused that (e.g. two people operated it simultaneously).  
May I set up an independent environment that only I can operate it? How?

pattacini, 2019-11-13:
On the one hand, Gitpod is tremendously helpful when we aim to avoid the installation burden; on the other hand, you cannot rely on Gitpod Online to control a physical device like a real robot.

However, Gipod makes use of [Eclipse Theia][1] and thus one can install locally [Theia on Docker technology][2]. From that point, you would need to acquire an extensive knowledge of Docker in order to set up a complete system along with a physical device.

When you spawn a Gitpod workspace, you are the unique owner, unless you decide to share it with a peer. Therefore, I believe the glitches you've been experiencing are normally due to the usage of a free online service.

I didn't get that you work with a real iCub 🤖 What is your lab?

Bear in mind that if your final aim is to go to the robot, you ought to install your system properly by following the instructions available at http://wiki.icub.org/wiki/ICub_Software_Installation.

[1]: github.com/eclipse-theia/theia
[2]: https://github.com/theia-ide/theia-apps#theia-docker

YANGGANG1971, 2019-11-14:
The University of Tokyo, Yukie Nagai
Yes, our final aim is to control the real iCub. 
May I ask you a question? 
Most of the programs are designed to simulate?  For a simulation project, how to transfer from simulation to real control?

pattacini, 2019-11-14:
I will try to formulate an answer on a different issue since the discussion is getting a bit off-topic wrt the original request.

closed 2019-11-14 by pattacini