vvv-school/vvv18 #59: How to run a compete demo using gitpod?

YANGGANG1971, 2019-11-12:
Following your suggestion, I try to use the “gitpod” tool.
First, I run the following codes in different console terminals on Ubuntu 16. 04.
>> yarp namespace /YANGGANG1971
>> yarpserver –write
>> yarp check
It seems OK. 
But, while I type (in the online environment using Google Chrome)
>>git checkout vvv18
>> ./test.sh
the webpage reminds me of the following errors,
CMake Error at CMakeLists.txt:8 (find_package):
By not providing "FindYARP.cmake" in CMAKE_MODULE_PATH this project has
asked CMake to find a package configuration file provided by "YARP", but
CMake did not find one.

Could not find a package configuration file provided by "YARP" with any of
the following names:

YARPConfig.cmake
yarp-config.cmake

Add the installation prefix of "YARP" to CMAKE_PREFIX_PATH or set
"YARP_DIR" to a directory containing one of the above files. If "YARP"
provides a separate development package or SDK, be sure it has been
installed.

-- Configuring incomplete, errors occurred!
See also "/workspace/tutorial_cartesian-interface/smoke-test/build/build-code/CMakeFiles/CMakeOutput.log".
 

 

May I need to install the yarp for gitpod? If yes, how to install yarp?
 If no, how to run a compete project using the gitpod? 
Could you give me a demo?

pattacini, 2019-11-12:
The environment we have made available through Gitpod is self-contained, meaning that you don't have to install anything.

By clicking on https://gitpod.io/#https://github.com/pattacini/technical-evaluation you'll be presented with the Gipod IDE, after a couple of minutes of wait ⌛️

From that point, just follow the [tutorial][1] I have prepared for you.

[1]: https://github.com/vvv-school/vvv18/files/3836461/tutorial-gitpod.zip

YANGGANG1971, 2019-11-13:
It does work while using the codes from https://github.com/vvv-school/tutorial_cartesian-interface.git

But when I cloned other codes, e.g. from https://github.com/vvv-school/tutorial_yarpmanager
Errors occurred as follows.
[image]
Besides, when I want to compile the c++ files, in gitpod, do we need to type the following?
>> cmake ../
>>make 
>>make install
Are all the operations/commands the same with the previous (Ubuntu terminal, CMake and YARP)

pattacini, 2019-11-13:
Try to use `https` instead of `ssh`:
```sh
$ git clone https://github.com/vvv-school/tutorial_yarpmanager.git
```

YANGGANG1971, 2019-11-13:
Thank you. You are right. Indeed, It was caused by SSH.

closed 2019-11-13 by pattacini