vvv-school/vvv18 #60: How to change the assignments code to make it work with the physical robot

pattacini, 2019-11-14:
May I ask you a question? 
Most of the programs are designed to simulate?  For a simulation project, how to transfer from simulation to real control?

_Originally posted by @YANGGANG1971 in https://github.com/vvv-school/vvv18/issues/58#issuecomment-553674423_

pattacini, 2019-11-14:
In general, it very much depends on the single assignment whether it is possible/doable to make it run along with the physical robot.

In particular, you can easily run [`tutorial_cartesian-interface`][1] on the iCub by undertaking the following adjustments:
- Instead of launching the modules in the accompanying application script, you have to rely on a [different XML][2], shipped with the `icub-main` repo.
- Replace the line https://github.com/vvv-school/tutorial_cartesian-interface/blob/master/src/main.cpp#L67 within the tutorial with:
  ```c++
  option.put("remote", "/icub/cartesianController/left_arm");
  ```

Other assignments that can be run on the real robot are:
- [`assignment_make-it-roll`][3]
- [`assignment_grasp-it`][4]

However, the underlying complexity does increase quite a lot and, most importantly, there's the need for having a supervisor aside, given that the porting to the iCub is not documented in those repositories.

[1]: https://github.com/vvv-school/tutorial_cartesian-interface
[2]: https://github.com/robotology/icub-main/blob/master/app/iCubStartup/scripts/iCubStartup.xml.template
[3]: https://github.com/vvv-school/assignment_make-it-roll
[4]: https://github.com/vvv-school/assignment_grasp-it

YANGGANG1971, 2019-12-04:
The installation for a laptop is as follows.
http://wiki.icub.org/wiki/ICub_server_laptop_installation_instructions
It seems that it only provides YARP > 2.3.23 and iCub > 1.1.13 software versions.
But the needed YARP and iCub in the tutorials are newer than the above versions.
When I want to transplant these codes from tutorials to the physical robot, does this affect the transplant?

pattacini, 2019-12-04:
Hi @YANGGANG1971 

Those instructions should be fine right because they hold for software versions newer than a given tag. Nonetheless, the page you pointed out refers to the setup of the robot laptop; as I suppose your robot infrastructure is already prepared, you should only update the software.

YANGGANG1971, 2019-12-05:
The main issue is that iCub has a lot of demos and can work well. I am afraid that once I updated, the demos are invalid.

pattacini, 2019-12-05:
I think this [dilemma][1] holds also in our case 😃 

[1]: https://billbennett.co.nz/operating-system-upgrade-dilemma

closed 2019-12-08 by pattacini