vvv-school/vvv18 #61: NoVNC often failed to connect server!!!

YANGGANG1971, 2019-11-24:
Using Gippod, "failed to connect to server" often happened and let me become speechless. Is it common while you used the Gitpod?
 
[image]

pattacini, 2019-11-24:
Hi @YANGGANG1971 

Nope, it's not common at all, at least in my experience. I can give you a few hints on this:
- Are you sure that you didn't close any terminal within Gitpod? One of the terminals opened up at startup is used to run the `noVNC` service. If you see this error, you could try to rerun the service by typing in a new shell `start-vnc-session.sh`.
- Instead of relying on `noVNC` you might consider installing a VNC client like [VNC Viewer][1] to then connect to the port `5901`.
- As a last resort, you can pull and run the docker image locally to your computer. First, install [docker][2], then follow these [instructions][3].

Hope this helps out

[1]: https://www.realvnc.com/en/connect/download/viewer/
[2]: https://phoenixnap.com/kb/how-to-install-docker-on-ubuntu-18-04
[3]: https://github.com/vvv-school/vvv-school.github.io/tree/master/dockerfiles

YANGGANG1971, 2019-11-25:
Great！ It works. Thank you.

pattacini, 2019-11-25:
⚠️ Consider that there's also a timeout of 30 minutes of inactivity. Once the timeout expires, you ought to restart the workspace.

closed 2019-11-25 by pattacini